microsoft/maker.js #521: Possible bug when aligning models

valeriu-balaban, 2022-03-14:
My goal is to programmatically align two text objects such that they are centered and the design has a height of 3 inches.

To do that, I run the code below in the playground. The main alignment happens on the line with `this.models.txtLine2.origin = [mName.width / 2 - mLine2.width / 2, mName.height - designHeight];` where `designHeight = 3`. However, the result is a design which has the height slightly larger that 3 inches.

Please correct me if I'm doing something wrong. Otherwise, I think there is a bug somewhere that messes up the alignment.

```
var makerjs = require('makerjs');

function TestDesign(font) {
  
    // Design constants for now
    var R = 27;
    var thetaDeg = 11;
    var designHeight = 3;
    var fontSizeName = 1.5;
    var fontSizeLine2 = 1;

    // Shortcuts
    var measurements = makerjs.measure.modelExtents;

	this.units = makerjs.unitType.Inch;
  
    this.models= {
      txtName: new makerjs.models.Text(font, '.' + "Millie's" + '.', fontSizeName),
      txtLine2: new makerjs.models.Text(font, "Toy Bin", fontSizeLine2, {combine: true})
    };

    makerjs.model.originate(this.models.txtName);
    makerjs.model.zero(this.models.txtName);
    makerjs.model.originate(this.models.txtLine2);
    makerjs.model.zero(this.models.txtLine2);

    var mName  = measurements(this.models.txtName);
    var mLine2 = measurements(this.models.txtLine2);

    // Center text on line 2
    // makerjs.model.move(design.models.txtLine2, [mName.width / 2 - mLine2.width / 2, mName.height - designHeight]);
    this.models.txtLine2.origin = [mName.width / 2 - mLine2.width / 2, mName.height - designHeight];

    console.info(measurements(this.models.txtName)); 
    console.info(measurements(this.models.txtLine2));

    makerjs.model.originate(this);
    makerjs.model.zero(this);

    console.info(measurements(this));
    console.info(this);
}

TestDesign.metaParameters = [
    { title: "font", type: "font", value: '*' }
];

module.exports = TestDesign;
```

valeriu-balaban, 2022-03-14:
The problem looks like is with the Text model. If I change the `models` property to use `Square` instead of `Text`:

```
    this.models= {
      txtName: new makerjs.models.Square(fontSizeName),
      txtLine2: new makerjs.models.Square(fontSizeLine2)
    };
```

results in the two models being aligned correctly.

danmarshall, 2022-03-22:
Hi @valeriu-balaban , it may be the `y` character dropping below the x-axis which causes the discrepancy.
Here's the changes I made to your code to get this result, as you can see is clamped to 3 inches:

[image]

```js
//center both text lines horizontally
   this.models.txtLine2.origin[0] = mName.width / 2 - mLine2.width / 2;

//move top line to align at 3 inches
  makerjs.model.move(this.models.txtName, [0, designHeight - mName.high[1]]);
```

valeriu-balaban, 2022-03-23:
Thanks a lot.

My confusion was that I expected the origin to be the bottom left corner of the text, however, for text it is the leftmost point on the text baseline.

closed 2022-03-23 by valeriu-balaban