microsoft/maker.js #528: toPdf() doesn't display the whole model content.

Sarthak0702, 2022-05-04:
On exporting to pdf vis toPDF() complete model is not displayed on the one page pdf generated, with is zoom/cropped in.

 dxf pic (actual Model)
[image]

pdf pic (cropped Model)
[image]

As u can see only this p[art of the actual model is displayed in the pdf
[image]

Here is the `iModel ` json that was exported: https://gist.github.com/Sarthak0702/640d5deecf363cae36f3ccadadeb626b

Code Used:
```js
exportToPDF(iModel: MakerJs.IModel){
  const doc: PDFKit.PDFDocument = new PDFDocument();
  const stream = doc.pipe(BlobStream());
  MakerJs.exporter.toPDF(
      doc, 
      iModel,
      {
        units: MakerJs.unitType.Millimeter
      }
  );
  doc.end();
  return stream;
}
```

danmarshall, 2022-05-04:
Hello, I think it's a scale issue. I was able to fit on one page by:

```js
var makerjs = require('makerjs');

var models = your_json;

var x = { models: models, units:'mm' };
makerjs.model.scale(x, 0.1)
module.exports = x;
```

[image]

Without scaling, the model appears as:
[image]

Each grid square is 100mm. A PDF page is about 216mm wide, so without scaling your model would take up about 10 pages wide.

Sarthak0702, 2022-05-04:
Thanks for this. I assumed that the exporter would automatically scale the content to fit it in one page. This dynamic scaling worked for me.

```js
let size = MakerJs.measure.modelExtents(iModel);
let xScale=size.width/216;
let yScale=size.height/280;
let scale=Math.max(xScale,yScale);
doc.scale(scale);
```

danmarshall, 2022-05-04:
No problem. The exporter doesn't automatically scale so that you can create a PDF in the exact coordinate space, for example if you are creating a ruler.

Sarthak0702, 2022-05-04:
Makes sense!. Closing issue :).

closed 2022-05-04 by Sarthak0702