microsoft/maker.js #532: Text intersection opposed to union of childrenOnPath

lachlansmith, 2022-06-25:
Hi @danmarshall thanks for pointing me here. I understand the use case for maker js is to draw outlines and that fill happens post exportation. That's what I've done. However, due to the union of glyph parts the fill results in white spaces. Clearly demonstrated by an A. 

[image]

[image]

My approach to resolving this would be to intersect the parts of the glyph, rather than default union. This doesn't seem to be exposed on ```MakerJs.models.Text```. I plan to use ```MakerJs.model.combineIntersection``` on a glyphs child models. I will be trying this soon but it seems excessive. Am I missing something?

danmarshall, 2022-06-25:
It appears the A character in this font is created with 2 overlapped closed shapes. This is pretty unusual for a font. Typically the author would ensure the character is a union. It’s ok to have multiple closed geometriessuch as an O or “I” but the assumption is they’re not overlapping. 

So you’ll need to take the A model and find the chains in it. Then convert the chains to models, then combine the models with a union. I’m mobile right now but you should be able to find these functions in the documentation.

lachlansmith, 2022-06-27:
Hi @danmarshall thanks for getting back to me. I managed to resolve the above. I've now encountered a separate issue related to chains. The ```toPDF``` method generates path information by finding chains. 
```
MakerJs.model.findChains(scaledModel, function (chains, loose, layer) {
    chains.map(function (chain) {
        if (chain.links.length > 1) {
            var pathData = exporter.chainToSVGPathData(chain, offset);
            doc.path(pathData).stroke(opts.stroke);
        }
...
```
Post exportation using fill with either pdfkit or pdf-lib results in any outer chain filling over any inner chain. 
[image]

As I understand it makerjs is for outlines, and fill is out of scope. However, I'm unaware of any other javascript pdf libraries, and noticed that ```toPDF``` likely doesn't support fill for this reason. Generating SVG path data that pdfkit supports seems like an easier option that correcting pdfkit's path/fill methods. My next approach is to use opentype's font.getPaths and makerjs importer with childrenOnPath to use path data supported by pdfkit. Is there a better solution here?

danmarshall, 2022-06-27:
Fill is not completely out of scope, because Maker.js is aware of [winding order](https://svgwg.org/svg2-draft/painting.html#WindingRule). See https://maker.js.org/docs/working-with-chains/#Find%20multiple%20chains

This might be as simple as adding the options `{ contain: { alternateDirection: true } }`  to the .findChains() call within .toPDF(). True that I didn't consider fill in `IPDFRenderOptions`, but it could (and probably should) be added.

lachlansmith, 2022-07-01:
Hi @danmarshall thanks for the suggestion. I've tried combinations of fill rules and ```{ contain: { alternateDirection: true } }``` to no avail. You've left a TODO for fill support. Perhaps [this](https://github.com/microsoft/maker.js/blob/6b31f30a8251c2d5ee34d21ac028e5edcf407475/packages/maker.js/src/core/pdf.ts#L72) is already supported elsewhere?

```
//TODO use only chainToSVGPathData instead of circle, so that we can use fill
```

Ideally, I'd like to use toSVGPathData and pdf-lib instead of pdfkit.

danmarshall, 2022-07-02:
Hi @lachlansmith , it makes sense, after thinking about it further, the existing code does not account for the nesting. Since the call you referenced uses [chainToSVGPathData](https://github.com/microsoft/maker.js/blob/6b31f30a8251c2d5ee34d21ac028e5edcf407475/packages/maker.js/src/core/svg.ts#L78) which does not descend into any [nested/contained chains](https://maker.js.org/docs/working-with-chains/#Find%20multiple%20chains). I think it *should* work if the function would crawl the [.contains](https://maker.js.org/docs/api/interfaces/makerjs.ichain.html#contains) recursively and return a union of all the svg.

lachlansmith, 2022-07-05:
Hi @danmarshall thanks for your assistance. I managed to resolve the above by changing the fill rule from the default evenodd to nonzero.
```
const path = MakerJs.exporter.toSVGPathData(model, {
    fillRule: 'nonzero',
});
```

lachlansmith, 2022-07-05:
Hi @danmarshall changing the fill rule wasn't the solution here. With almost all fonts I try there always seems to be at least one character not filling correctly.

```
const path = MakerJs.exporter.toSVGPathData(model, {
    fillRule: 'evenodd',
});
```
This generates:
[image]

```
const path = MakerJs.exporter.toSVGPathData(model, {
    fillRule: 'nonzero',
});
```
This generates:
[image]

At first, I thought I had this backwards. The documentation you linked suggests those two should be opposite. 
[image]

The toSVGPathData has likely been around for some time so forgive me when I ask is something going wrong here? Every font I try seems to break.

danmarshall, 2022-07-05:
For context - this is only happening in the PDF output, correct?

lachlansmith, 2022-07-05:
This is happening with toSVGPathData getting parsed to either drawSvgPathData from pdf-lib or doc.path().fill() from pdfkit.

lachlansmith, 2022-07-05:
Example fonts:
[Archive.zip](https://github.com/microsoft/maker.js/files/9050238/Archive.zip)

danmarshall, 2022-07-05:
Also seems to be happening with some Google fonts: https://github.com/danmarshall/google-font-to-svg-path/issues/30

danmarshall, 2022-07-06:
Unfortunately, some fonts are degenerate, regarding each glyph having a closed geometry which is non-overlapping. Let's look at Bodoni Moda:

[image]

https://danmarshall.github.io/google-font-to-svg-path/?font-select=Bodoni+Moda&font-variant=regular&input-union=false&input-filled=false&input-kerning=true&input-separate=false&input-text=Verb+1234&input-bezier-accuracy=&dxf-units=cm&input-size=300&input-fill=%23000&input-stroke=%23000&input-strokeWidth=0.25mm

Notice that with the glyphs `r`, `1`, `2` and `3` you could use the strategy of combining sub-geometries. But with the `e` and `4`, there is a singular geometry which *is* closed but is self-overlapping.

This wasn't common some time ago, but now there seem to be many fonts assembled this way. I don't currently have a solution. One idea that springs to mind is to create a process to compare 2 bitmaps of a shape, one rendered with even-odd and the other with nonzero. If they differ, the shape is degenerate. 🤔

danmarshall, 2022-07-06:
Fonts seem to be consistent within themselves. Perhaps you can store the fill rule associated with each font, then apply correspondingly?

lachlansmith, 2022-07-06:
How does opentype.js approach this? The output of ```const path = font.getPath("Verb 1234", 0, 0, 32).toPathData(3)``` consistently fills correctly with both pdf-lib & pdfkit. 
```
    const doc = await PDFDocument.create();

    const font = fs.readFileSync('./examples/font.ttf');

    const f = opentype.parse(font.buffer);
    const path = f.getPath('Verb 1234', 0, 0, 48).toPathData(3);

    const page = doc.addPage([300.5, 300.5]);

    page.drawSvgPath(path, {
        x: 0,
        y: 150.25,
        color: rgb('#000000'),
    });

    fs.writeFileSync('output.pdf', await doc.save());
```
[image]

Unfortunately, I'm not well versed enough with vector graphics. I'd like to be. I'm going to have a dig around and see what conclusions can be drawn from the difference between makerjs output and opentype.js output.

lachlansmith, 2022-07-18:
Hi @danmarshall thanks for queueing this work up. What does MakerJs' timeline look like? MakerJs' covers all of my other basis. I'd much rather move forward when support is in place. Not asking for a date, but a general idea of when I can revisit this would be fantastic.

danmarshall, 2022-07-18:
@lachlansmith the timeline is quite long right now, as this is my fun side project and gets delayed for higher priorities. I have one pull request that is almost 2 years old now 🙁

kasterlod, 2025-12-30:
hey @danmarshall! so a long time :D can you point me to your PR? Looks like I have a similar issue. Example below

[image]

this is polish "ę" and noone of combine method works. My last idea is to someohow extract combine from text methos which works when letter follows next letter and reuse it somehow here to get rid of it. Best regards!

danmarshall, 2026-01-05:
Hi @kasterlod :D i'm wondering if the glyph and the e character are one model? if so you might have luck if you can separate them into  2 models. There is definitely a simple algorithm in the text combine that assumes (wrongly) that each character is already a closed geometry.