microsoft/maker.js #546: model outline not working with some values.

tarikjabiri, 2022-12-30:
Hi

I am using model outline but in some values is not working

[image]

Example to reproduce: 

```js
var MakerJs = require('makerjs');

function Example(radius, width, inset, rcorner) {
  if (arguments.length === 0) {
    var defaultValues = MakerJs.kit.getParameterValues(QuarterCircle);
    radius = defaultValues.shift();
    width = defaultValues.shift();
    inset = defaultValues.shift();
    rcorner = defaultValues.shift();
  }

  var right = new MakerJs.paths.Line([radius, 0], [radius + width, 0]);
  var barc = new MakerJs.paths.Arc(MakerJs.point.zero(), radius + width, 0, 90);
  var left = new MakerJs.paths.Line([0, radius + width], [0, radius]);
  var sarc = new MakerJs.paths.Arc(MakerJs.point.zero(), radius , 0, 90);

  // Corners if rounded
  var first = MakerJs.path.fillet(sarc, right, rcorner);
  var second = MakerJs.path.fillet(right, barc, rcorner);
  var third = MakerJs.path.fillet(barc, left, rcorner);
  var fourth = MakerJs.path.fillet(left, sarc, rcorner);

  var top = {
    paths: {
      first: first, right: right, second: second,
      barc: barc, third: third, left: left, fourth: fourth,
      sarc: sarc
    }
  };
  
  top.layer = 'red';
  
  var eouter = MakerJs.model.outline(top, inset, 1, true);
  var einner = MakerJs.model.outline(eouter, 15, 1, true);
  var router = MakerJs.model.outline(eouter, 1, 1, true);
  var rinner = MakerJs.model.outline(einner, 1, 1);
  
  eouter.layer = 'aqua';
  einner.layer = 'aqua';
  router.layer = 'yellow';
  rinner.layer = 'yellow';

  this.models = {
    top: top,
    eouter: eouter,
    einner: einner,
    router: router,
    rinner: rinner
  };

}

Example.metaParameters = [
  { title: "radius", type: "range", min: 100, max: 2000, value: 250 },
  { title: "width", type: "range", min: 100, max: 2000, value: 250 },
  { title: "inset", type: "range", min: 1, max: 20, value: 2 },
  { title: "rcorner", type: "range", min: 0, max: 100, value: 0 }
];

module.exports = Example;
```

Regards

danmarshall, 2022-12-31:
Thanks for the great repro code. The boolean operations code does have a lot of issues unfortunately. Sometimes scaling your model up helps (then scale it down afterwards). I've been working on a fix for a while but haven't perfected it.

tarikjabiri, 2022-12-31:
Hi

the boolean operations are working ok for my needs, the poblem is the outline, offsetting is not working in a specific values.
I think the problem is related to the floating numbers, the precision.

regards

danmarshall, 2023-01-03:
A little background: outlines are created using a series of [combineUnion](https://maker.js.org/docs/advanced-drawing/#Combining%20with%20Boolean%20operations) boolean operations. 

You're correct that the issue is related to binary floating point. My hope is to use a library like https://github.com/mourner/robust-predicates to correct the issue.