microsoft/maker.js #549: childrenOnChain Issue with cloneToRow

noobd3v, 2023-01-20:
I am using childrenOnChain with cloneToRow to load rects on a svg path but its not working all the time.
here is one example where it is loading some elements on the path but others are on left corner 

here margin was 8
[image]

When I decreased the margin and increased the count for some of the paths it was working for the paths which had more than 1 rect.
margin set to 4.
[image]
```

here is the code I am using.
let model: any = {
    models: {
    }
};
const rect = new makerjs.models.Rectangle(1, 1);
const margin = 8;
for (let i = 0; i < pathsArray.length; i++) {
    let pathModel = makerjs.importer.fromSVGPathData(pathsArray[i]);
    const chain = makerjs.model.findSingleChain(pathModel);
    const count = Math.ceil(chain.pathLength / margin);
    const row = makerjs.layout.cloneToRow(rect, count, margin);
    makerjs.layout.childrenOnChain(row, chain, 0.5, false, true);

    model =
    {
        ...model,
        [`path${i}`]: pathModel,
        [`row${i}`]: row,
    };
}
makerjs.exporter.toSVG(model);

```
Where Paths array is 
```

0:
'M4.388292781978905-8.821125724904025q0.07606288608417966-0.09622641963270162 0.14920757715050303-0.18238822679407818M4.537500359129408-9.003513951698103q0.058867364858572024-0.06934363198613447 0.11608955486124994-0.13245124780174677M4.653589913990658-9.13596519949985q0.14068004700300563-0.1551492936354304 0.2851772936329118-0.2865029636213432M4.938767207623569-9.422468163121193q0.06468627662006199-0.058802364958078 0.1292900342716683-0.11202012612125145M5.068057241895238-9.534488289242445q0.05…15762743870691764M9.24804263803006-10.103429891977823q-0.05886339243129157-0.0067383691316873495-0.12423976181915286-0.012743065228416839M9.123802876210908-10.11617295720624q-0.06039229360669651-0.005546918146238866-0.12811999480373437-0.010630087690483236M8.995682881407173-10.126803044896723q-0.007785411944610132-0.0005843187969887254-0.01568515515493374-0.0011639296326926285M8.97999772625224-10.127966974529416q-0.07069949140058185-0.005187281434794855-0.15280388040867265-0.010147305380840521'
1:
'M10.700016025180416-9.680782176187579q0.11545707409388761-0.043038139751015336 0.22546796854569884-0.08127545083821275M10.925483993726115-9.762057627025792q0.11446435228373808-0.03978523280333057 0.22457768623655205-0.07491145513202824M11.150061679962667-9.83696908215782q0.11335953819184574-0.03616176350914557 0.2237090260278034-0.067894689939072M11.37377070599047-9.904863772096892q0.11251632202182549-0.03235603752137273 0.22307261834584224-0.06044449085277748M11.596843324336312-9.9653082629496….011339137626638518M12.226306610290374-10.094286934283089q0.021426903428139354-0.003343918360576126 0.0429402380173034-0.006563043111807332M12.269246848307677-10.100849977394896q0.05666765452841638-0.008479403717288392 0.1140961118166004-0.016116169587570894M12.383342960124278-10.116966146982467q0.05709533527142696-0.007592467712742135 0.11521869445388866-0.014388553496196721M12.498561654578166-10.131354700478663q0.04227022286904436-0.004942454543996888 0.08524712707566096-0.009482997661024228'
2:
'M7.158830241900957-5.155220531669375l0.08651814147600945 0.08614681898040288M7.2453483833769665-5.069073712688972q0.11334078138140757 0.11285434025959873 0.2145892275426995 0.2353377622685855M7.459937610919666-4.833735950420387l0.7976220656827442 0.9649084383879547M8.25755967660241-3.868827512032432l0.7593929768632854 0.9186615101987199M9.016952653465696-2.950166001833712c0.03777989616130917 0.04570352573717207 0.07600929919960464 0.09103507941116229 0.11341143493743822 0.13705120620363154M9.13…40307433092485M12.845794936566971-0.498148142585295q0.012122327364622265 0.0024549683061168004 0.024291178833991012 0.0048779558425790515M12.870086115400962-0.49327018674271594q0.061395373472212356 0.012224672566187866 0.12407571989766453 0.023654638212599055M12.994161835298627-0.4696155485301169q0.07340768338863413 0.013386130536349572 0.14884032654104296 0.025729721825329566M13.14300216183967-0.4438858267047873q0.02581646130035864 0.00422453507529269 0.051893553848360696 0.008330931777796369'
3:
'M9.709041482919417-1.6217314064376906q-0.06167855681364287 0.07131690256466494-0.12442671272118133 0.14036139548448356M9.584614770198236-1.481370010953207q-0.056301749972723414 0.061951235397151194-0.11392962831520492 0.12258414605350354M9.470685141883031-1.3587858648997035q-0.10948550681221647 0.11519467909713033-0.20429253776645062 0.20615522412516896M9.26639260411658-1.1526306407745346q-0.050832549051385456 0.04877018423992885-0.10054727990216605 0.09366931472561046M9.165845324214414-1.05896…286929976M7.792855883054891-0.4213220494702693q-0.015946793687317218 0.0015532349168380088-0.032492325625009144 0.0030443439135844352M7.760363557429882-0.41827770555668486q-0.05262157866293293 0.004742338273727498-0.11309472675370813 0.009003585119765678M7.647268830676174-0.4092741204369192q-0.053995845323067115 0.0038048230139549055-0.11758507284731934 0.007458368767746526M7.529683757828854-0.40181575166917266q-0.05528037607352854 0.0031761572067283694-0.12293465313728191 0.006527859431192495'
4:
'M5.9831512893001095-4.266625974471071q-0.08589490607482553 0.06492056854492567-0.1559690147677344 0.14375581430526996M5.827182274532375-4.122870160165801l-0.5269040323187459 0.5927811235102878M5.300278242213629-3.5300890366555135q-0.39757215467461204 0.4472793033814839-0.43448076684302617 0.4888024788487817M4.865797475370603-3.041286557806732q-0.20800309711790366 0.23400904536745148-0.2680265319485535 0.30422875623937795M4.5977709434220495-2.737057801567354q-0.08516035087214124 0.09962667469573…014 0.006913780440570894M4.762043196315336-0.4233823632756026q0.06005432215478379 0.010161145447568465 0.12550673863763429 0.019354570863440668M4.88754993495297-0.4040277924121619q0.06128452789053895 0.008608005121647888 0.12849319678927973 0.01653545615815738M5.01604313174225-0.38749233625400453q0.06341660267230242 0.00748016618725017 0.1333172893168415 0.01449722472988807M5.149360421059091-0.37299511152411646q0.029864169620140046 0.0029979480404455083 0.06109506939423337 0.005930562656918881'
5:
'M3.541012851460431-0.9531678940844294q-0.06707982599235685 0.02296578351603451-0.14100222271231377 0.04734644701847224M3.400010628748117-0.9058214470659571q-0.04999273632535539 0.016488319318596067-0.2281780719912847 0.07445133833994799M3.1718325567568324-0.8313701087260091q-0.11912598178157374 0.03875123349593834-0.2220607834758157 0.07024672743385618M2.9497717732810167-0.761123381292153q-0.11069384688003359 0.033869569144658995-0.21819699707160511 0.06425522850793508M2.7315747762094116-0.6968…77 0.04645426702281075M2.093692327706382-0.5399884840630638q-0.10455553659081951 0.021774270733543433-0.20863792005420345 0.040814628882548776M1.8850544076521785-0.499173855180515q-0.10387302436915924 0.01900205894789997-0.2073793525729717 0.03530043483813622M1.6776750550792068-0.4638734203423788q-0.10342013848954879 0.01628480423345624-0.20656404999905287 0.029884614381296526M1.471111005080154-0.43398880596108225q-0.09347432648828646 0.012324848605540839-0.1867890938819987 0.02245349088005394'
6:
'M3.035000000000082-9.462529612545636c-0.10315405490980512-0.04889068553842435-0.20543484528131417-0.09959059171439577-0.30840620280033315-0.14885978785910403M2.7265937971997487-9.61138940040474q-0.13268468986914783-0.06348627587390254-0.2442711280751575-0.11274762669517457M2.482322669124591-9.724137027099914q-0.06637590478131905-0.029302545937305524-0.12935403528744205-0.05543018028639857M2.352968633837149-9.779567207386313q-0.06784905419803344-0.028148426521578784-0.13318165235385537-0.0532065…085709637M1.5200743219848403-10.034706739878201q-0.06644853637599413-0.012726486123822767-0.14543130636640678-0.025119561346357955M1.3746430156184335-10.05982630122456q-0.07997556718103183-0.012548853631745516-0.14239602368901827-0.021241169040074226M1.2322469919294152-10.081067470264633q-0.0028808988690838255-0.00040117748299906-0.005747743088176049-0.0007976682074346542M1.2264992488412392-10.081865138472068q-0.013805198328857626-0.0019092886351899097-0.02733213255791611-0.0037164045930406786'

```

danmarshall, 2023-01-20:
Hello, in your image, I see that 3 paths converge at one point:
[image]

I'm wondering if this is failing the chain-finding algorithm. Can you remove the convergence and try again?

noobd3v, 2023-01-21:
Not sure that’s the case because I am running the chain algorithm separately on each path. Also it works for that path of margin is less and count is more than 1.
For all the paths that are failing the count is 1 with same margin.
But let me try

noobd3v, 2023-01-21:
> Hello, in your image, I see that 3 paths converge at one point: [image]
> 
> I'm wondering if this is failing the chain-finding algorithm. Can you remove the convergence and try again?

Yep still not working after removing the path.

[image]

noobd3v, 2023-01-23:
@danmarshall 
I also tried placing rectangles without chilldrenOnChain but its placing it out of the path. 
I am getting the location of placement from the path and using that to move the rectangle at the location.
the location is based on a object from pathModel. 
```
makerjs.model.move(rectangle, location);
model =
    {
        ...model,
        [`path${i}`]: pathModel,
        [`row${i}`]: rectangle,
    };
```

[image]

danmarshall, 2023-01-23:
Hello @noobd3v, I'm having trouble using your path data. Let's simplify to one simple scenario so we can repro.
The paths array pathdata you provided, I'm not able to render in other online tools such as https://yqnn.github.io/svg-path-editor/ or  https://svgomg.net/
Can you provide just one path data that is renderable?

noobd3v, 2023-01-24:
Hi @danmarshall, Sorry my bad some of the paths were truncated while I copied . Please refer below the set of paths they all are working now

0:
'M4.388293-8.821126Q4.464356-8.917352 4.5375-9.003514M4.5375-9.003514Q4.596368-9.072858 4.65359-9.135965M4.65359-9.135965Q4.79427-9.291114 4.938767-9.422468M4.938767-9.422468Q5.003453-9.481271 5.068057-9.534488M5.068057-9.534488Q5.12793-9.583809 5.187824-9.628409M5.187824-9.628409Q5.242993-9.669489 5.298286-9.706644M5.298286-9.706644Q5.348987-9.740712 5.399916-9.771562M5.399916-9.771562Q5.446496-9.799777 5.493401-9.825381M5.493401-9.825381Q5.536285-9.84879 5.579583-9.870094M5.579583-9.870094Q5.619236-9.889605 5.659385-9.907422M5.659385-9.907422Q5.696284-9.923798 5.733751-9.93881M5.733751-9.93881Q5.768357-9.952675 5.803595-9.965437M5.803595-9.965437Q5.836335-9.977294 5.869764-9.988249M5.869764-9.988249Q5.901027-9.998495 5.933027-10.007996M5.933027-10.007996Q5.991647-10.025402 6.053464-10.040515M6.053464-10.040515Q6.10966-10.054255 6.169336-10.066306M6.169336-10.066306Q6.224756-10.077498 6.283913-10.087384M6.283913-10.087384Q6.295412-10.089306 6.307069-10.091181M6.307069-10.091181Q6.352033-10.098414 6.399582-10.104994M6.399582-10.104994Q6.456619-10.112887 6.518082-10.119935M6.518082-10.119935Q6.57682-10.126671 6.640692-10.132758M6.640692-10.132758Q6.700214-10.138431 6.766135-10.143722M4.388293-8.821126Q4.399829-8.7978 4.41173-8.774369M4.41173-8.774369Q4.45334-8.692443 4.499166-8.609669M4.499166-8.609669Q4.543648-8.529323 4.591445-8.449364M4.591445-8.449364Q4.640969-8.366514 4.693366-8.285233M4.693366-8.285233Q4.749453-8.198227 4.808287-8.113867M4.808287-8.113867Q4.872417-8.021913 4.93946-7.933614M4.93946-7.933614Q4.975471-7.886185 5.012291-7.839854M5.012291-7.839854Q5.045116-7.798548 5.078582-7.758119M5.078582-7.758119Q5.090011-7.744311 5.881601-6.802823M5.881601-6.802823L6.664162-5.872074M6.664162-5.872074Q6.750616-5.769248 6.845624-5.673418M6.845624-5.673418L6.9589-5.559162M6.9589-5.559162L8.145691-6.463383M8.145691-6.463383Q8.217907-6.518405 8.27882-6.583845M8.27882-6.583845L8.825905-7.171586M8.825905-7.171586L9.102077-7.468282M9.102077-7.468282C9.241424-7.617984 9.382549-7.766027 9.520825-7.91673M9.520825-7.91673Q9.599154-8.002099 9.668271-8.081936M9.668271-8.081936Q9.70432-8.123577 9.738177-8.164068M9.738177-8.164068Q9.773035-8.205759 9.805768-8.246469M9.805768-8.246469Q9.839527-8.288456 9.871221-8.329645M9.871221-8.329645Q9.903971-8.372206 9.93471-8.414166M9.93471-8.414166Q9.96653-8.457604 9.996387-8.500659M9.996387-8.500659Q10.023246-8.539392 10.05587-8.590027M10.05587-8.590027Q10.085606-8.636182 10.113712-8.682708M10.113712-8.682708Q10.142701-8.730698 10.170128-8.779372M10.170128-8.779372Q10.198395-8.829536 10.225185-8.880749M10.225185-8.880749Q10.230051-8.890051 10.234872-8.899395M10.234872-8.899395Q10.257351-8.942962 10.27891-8.987548M10.27891-8.987548Q10.305728-9.043013 10.331304-9.100433M10.331304-9.100433Q10.337752-9.11491 10.344129-9.12953M10.344129-9.12953Q10.385597-9.22459 10.417641-9.309912M10.417641-9.309912Q10.420461-9.317421 10.423248-9.324953M10.423248-9.324953Q10.438711-9.36674 10.453201-9.409344M10.453201-9.409344Q10.47179-9.463999 10.488901-9.520365M10.488901-9.520365Q10.495469-9.541999 10.501831-9.563924M10.501831-9.563924Q10.508479-9.586835 10.51491-9.610086M10.51491-9.610086Q10.499723-9.621446 10.484562-9.632512M10.484562-9.632512Q10.429641-9.672601 10.374975-9.70889M10.374975-9.70889Q10.339958-9.732135 10.305-9.753853M10.305-9.753853Q10.287732-9.76458 10.270471-9.77494M10.270471-9.77494Q10.259619-9.781453 10.248768-9.787821M10.248768-9.787821Q10.210143-9.810491 10.171481-9.831358M10.171481-9.831358Q10.124935-9.856482 10.078198-9.879066M10.078198-9.879066Q10.034541-9.900162 9.990552-9.919123M9.990552-9.919123Q9.949629-9.936762 9.90824-9.95263M9.90824-9.95263Q9.869801-9.967368 9.830773-9.980649M9.830773-9.980649Q9.794505-9.992991 9.757537-10.00414M9.757537-10.00414Q9.723101-10.014527 9.687862-10.023937M9.687862-10.023937Q9.65491-10.032737 9.621065-10.040734M9.621065-10.040734Q9.589266-10.048249 9.556494-10.055098M9.556494-10.055098Q9.525541-10.061568 9.493542-10.067482M9.493542-10.067482Q9.434522-10.07839 9.370393-10.087667M9.370393-10.087667Q9.312205-10.096085 9.248043-10.10343M9.248043-10.10343Q9.189179-10.110168 9.123803-10.116173M9.123803-10.116173Q9.063411-10.12172 8.995683-10.126803M8.995683-10.126803Q8.987897-10.127387 8.979998-10.127967M8.979998-10.127967Q8.909298-10.133154 8.827194-10.138114'
1:
'M10.700016-9.680782Q10.815473-9.72382 10.925484-9.762058M10.925484-9.762058Q11.039948-9.801843 11.150062-9.836969M11.150062-9.836969Q11.263421-9.873131 11.373771-9.904864M11.373771-9.904864Q11.486287-9.93722 11.596843-9.965308M11.596843-9.965308Q11.652711-9.979502 11.708292-9.992663M11.708292-9.992663Q11.764119-10.005881 11.819804-10.018092M11.819804-10.018092Q11.875648-10.030337 11.9315-10.041602M11.9315-10.041602Q11.956776-10.0467 11.982078-10.051602M11.982078-10.051602Q12.01277-10.057548 12.043525-10.063211M12.043525-10.063211Q12.099612-10.073539 12.156044-10.082948M12.156044-10.082948Q12.191079-10.088789 12.226307-10.094287M12.226307-10.094287Q12.247734-10.097631 12.269247-10.10085M12.269247-10.10085Q12.325915-10.109329 12.383343-10.116966M12.383343-10.116966Q12.440438-10.124559 12.498562-10.131355M12.498562-10.131355Q12.540832-10.136297 12.583809-10.140838'
2:
[... cut ...] 083592-9.882644Q2.013848-9.906938 1.945132-9.928262M1.945132-9.928262Q1.878606-9.948905 1.812381-9.966971M1.812381-9.966971Q1.808672-9.967983 1.804962-9.968987M1.804962-9.968987Q1.734341-9.988103 1.663326-10.004486M1.663326-10.004486Q1.592254-10.020883 1.520074-10.034707M1.520074-10.034707Q1.453626-10.047433 1.374643-10.059826M1.374643-10.059826Q1.294667-10.072375 1.232247-10.081067M1.232247-10.081067Q1.229366-10.081469 1.226499-10.081865M1.226499-10.081865Q1.212694-10.083774 1.199167-10.085582'

danmarshall, 2023-01-25:
Hello, I'm going to close this issue since #550 is similar and [making sure count > 1](https://github.com/microsoft/maker.js/issues/550#issuecomment-1404167358) is probably the workaround.

noobd3v, 2023-01-26:
> As a workaround for now, can you ensure your calling code only calls layout with a count > 1?

well it is very much needed in some cases. That's why I was trying to figure out the alternates also with childrenOnPath. I was trying to place single rect on a path. And as alternate I also tried placing it on one of the coordinates from the path but the actual placement after exporting was different. 
For example these coordinates `[13.204878491338434, 7.632513965353765]` belong to one of the path
> 1:
> 'M10

But I am getting following result not sure what I am doing wrong. I also tried walking the path to see the offset but the offset was 0,0.
[image]

danmarshall, 2023-01-26:
Can you provide code for the repro?

noobd3v, 2023-01-27:
```
let pathsArray = [
            'M4.388293-8.821126Q4.464356-8.917352 4.5375-9.003514M4.5375-9.003514Q4.596368-9.072858 4.65359-9.135965M4.65359-9.135965Q4.79427-9.291114 4.938767-9.422468M4.938767-9.422468Q5.003453-9.481271 5.068057-9.534488M5.068057-9.534488Q5.12793-9.583809 5.187824-9.628409M5.187824-9.628409Q5.242993-9.669489 5.298286-9.706644M5.298286-9.706644Q5.348987-9.740712 5.399916-9.771562M5.399916-9.771562Q5.446496-9.799777 5.493401-9.825381M5.493401-9.825381Q5.536285-9.84879 5.579583-9.870094M5.579583-9.870094Q5.619236-9.889605 5.659385-9.907422M5.659385-9.907422Q5.696284-9.923798 5.733751-9.93881M5.733751-9.93881Q5.768357-9.952675 5.803595-9.965437M5.803595-9.965437Q5.836335-9.977294 5.869764-9.988249M5.869764-9.988249Q5.901027-9.998495 5.933027-10.007996M5.933027-10.007996Q5.991647-10.025402 6.053464-10.040515M6.053464-10.040515Q6.10966-10.054255 6.169336-10.066306M6.169336-10.066306Q6.224756-10.077498 6.283913-10.087384M6.283913-10.087384Q6.295412-10.089306 6.307069-10.091181M6.307069-10.091181Q6.352033-10.098414 6.399582-10.104994M6.399582-10.104994Q6.456619-10.112887 6.518082-10.119935M6.518082-10.119935Q6.57682-10.126671 6.640692-10.132758M6.640692-10.132758Q6.700214-10.138431 6.766135-10.143722M4.388293-8.821126Q4.399829-8.7978 4.41173-8.774369M4.41173-8.774369Q4.45334-8.692443 4.499166-8.609669M4.499166-8.609669Q4.543648-8.529323 4.591445-8.449364M4.591445-8.449364Q4.640969-8.366514 4.693366-8.285233M4.693366-8.285233Q4.749453-8.198227 4.808287-8.113867M4.808287-8.113867Q4.872417-8.021913 4.93946-7.933614M4.93946-7.933614Q4.975471-7.886185 5.012291-7.839854M5.012291-7.839854Q5.045116-7.798548 5.078582-7.758119M5.078582-7.758119Q5.090011-7.744311 5.881601-6.802823M5.881601-6.802823L6.664162-5.872074M6.664162-5.872074Q6.750616-5.769248 6.845624-5.673418M6.845624-5.673418L6.9589-5.559162M6.9589-5.559162L8.145691-6.463383M8.145691-6.463383Q8.217907-6.518405 8.27882-6.583845M8.27882-6.583845L8.825905-7.171586M8.825905-7.171586L9.102077-7.468282M9.102077-7.468282C9.241424-7.617984 9.382549-7.766027 9.520825-7.91673M9.520825-7.91673Q9.599154-8.002099 9.668271-8.081936M9.668271-8.081936Q9.70432-8.123577 9.738177-8.164068M9.738177-8.164068Q9.773035-8.205759 9.805768-8.246469M9.805768-8.246469Q9.839527-8.288456 9.871221-8.329645M9.871221-8.329645Q9.903971-8.372206 9.93471-8.414166M9.93471-8.414166Q9.96653-8.457604 9.996387-8.500659M9.996387-8.500659Q10.023246-8.539392 10.05587-8.590027M10.05587-8.590027Q10.085606-8.636182 10.113712-8.682708M10.113712-8.682708Q10.142701-8.730698 10.170128-8.779372M10.170128-8.779372Q10.198395-8.829536 10.225185-8.880749M10.225185-8.880749Q10.230051-8.890051 10.234872-8.899395M10.234872-8.899395Q10.257351-8.942962 10.27891-8.987548M10.27891-8.987548Q10.305728-9.043013 10.331304-9.100433M10.331304-9.100433Q10.337752-9.11491 10.344129-9.12953M10.344129-9.12953Q10.385597-9.22459 10.417641-9.309912M10.417641-9.309912Q10.420461-9.317421 10.423248-9.324953M10.423248-9.324953Q10.438711-9.36674 10.453201-9.409344M10.453201-9.409344Q10.47179-9.463999 10.488901-9.520365M10.488901-9.520365Q10.495469-9.541999 10.501831-9.563924M10.501831-9.563924Q10.508479-9.586835 10.51491-9.610086M10.51491-9.610086Q10.499723-9.621446 10.484562-9.632512M10.484562-9.632512Q10.429641-9.672601 10.374975-9.70889M10.374975-9.70889Q10.339958-9.732135 10.305-9.753853M10.305-9.753853Q10.287732-9.76458 10.270471-9.77494M10.270471-9.77494Q10.259619-9.781453 10.248768-9.787821M10.248768-9.787821Q10.210143-9.810491 10.171481-9.831358M10.171481-9.831358Q10.124935-9.856482 10.078198-9.879066M10.078198-9.879066Q10.034541-9.900162 9.990552-9.919123M9.990552-9.919123Q9.949629-9.936762 9.90824-9.95263M9.90824-9.95263Q9.869801-9.967368 9.830773-9.980649M9.830773-9.980649Q9.794505-9.992991 9.757537-10.00414M9.757537-10.00414Q9.723101-10.014527 9.687862-10.023937M9.687862-10.023937Q9.65491-10.032737 9.621065-10.040734M9.621065-10.040734Q9.589266-10.048249 9.556494-10.055098M9.556494-10.055098Q9.525541-10.061568 9.493542-10.067482M9.493542-10.067482Q9.434522-10.07839 9.370393-10.087667M9.370393-10.087667Q9.312205-10.096085 9.248043-10.10343M9.248043-10.10343Q9.189179-10.110168 9.123803-10.116173M9.123803-10.116173Q9.063411-10.12172 8.995683-10.126803M8.995683-10.126803Q8.987897-10.127387 8.979998-10.127967M8.979998-10.127967Q8.909298-10.133154 8.827194-10.138114',
            'M10.700016-9.680782Q10.815473-9.72382 10.925484-9.762058M10.925484-9.762058Q11.039948-9.801843 11.150062-9.836969M11.150062-9.836969Q11.263421-9.873131 11.373771-9.904864M11.373771-9.904864Q11.486287-9.93722 11.596843-9.965308M11.596843-9.965308Q11.652711-9.979502 11.708292-9.992663M11.708292-9.992663Q11.764119-10.005881 11.819804-10.018092M11.819804-10.018092Q11.875648-10.030337 11.9315-10.041602M11.9315-10.041602Q11.956776-10.0467 11.982078-10.051602M11.982078-10.051602Q12.01277-10.057548 12.043525-10.063211M12.043525-10.063211Q12.099612-10.073539 12.156044-10.082948M12.156044-10.082948Q12.191079-10.088789 12.226307-10.094287M12.226307-10.094287Q12.247734-10.097631 12.269247-10.10085M12.269247-10.10085Q12.325915-10.109329 12.383343-10.116966M12.383343-10.116966Q12.440438-10.124559 12.498562-10.131355M12.498562-10.131355Q12.540832-10.136297 12.583809-10.140838',
            'M7.15883-5.155221L7.245348-5.069074M7.245348-5.069074Q7.358689-4.956219 7.459938-4.833736M7.459938-4.833736L8.25756-3.868828M8.25756-3.868828L9.016953-2.950166M9.016953-2.950166C9.054733-2.904462 9.092962-2.859131 9.130364-2.813115M9.130364-2.813115Q9.19069 [... cut ...]
                    [`path${i}`]: pathModel,
                    [`row${i}`]: row,
                };
            }
            else
            {
                let rectClone= makerjs.model.clone(rect); rectClone.origin = [13.204878491338434, 7.632513965353765];
                    model =
                    {
                        ...model,
                        [`path${i}`]: pathModel,
                        [`row${i}`]: rectClone,
                    };
            }        
        }
        makerjs.exporter.toSVG(model);
```

noobd3v, 2023-01-27:
I see one problem with my code i.e. Arc is from a Bezier curve and that data is not taken in context while generating the starlight lines. But there s not a straighten method that will take full model context in arguments to provide the straighten arc. How can I straighten such arcs?

danmarshall, 2023-01-28:
Fix is coming:
[image]

Here's some simplified code:
```js
var makerjs = require('makerjs');

var pathsArray = [
  'M4.388293-8.821126Q4.464356-8.917352 4.5375-9.003514M4.5375-9.003514Q4.596368-9.072858 4.65359-9.135965M4.65359-9.135965Q4.79427-9.291114 4.938767-9.422468M4.938767-9.422468Q5.003453-9.481271 5.068057-9.534488M5.068057-9.534488Q5.12793-9.583809 5.187824-9.628409M5.187824-9.628409Q5.242993-9.669489 5.298286-9.706644M5.298286-9.706644Q5.348987-9.740712 5.399916-9.771562M5.399916-9.771562Q5.446496-9.799777 5.493401-9.825381M5.493401-9.825381Q5.536285-9.84879 5.579583-9.870094M5.579583-9.870094Q5.619236-9.889605 5.659385-9.907422M5.659385-9.907422Q5.696284-9.923798 5.733751-9.93881M5.733751-9.93881Q5.768357-9.952675 5.803595-9.965437M5.803595-9.965437Q5.836335-9.977294 5.869764-9.988249M5.869764-9.988249Q5.901027-9.998495 5.933027-10.007996M5.933027-10.007996Q5.991647-10.025402 6.053464-10.040515M6.053464-10.040515Q6.10966-10.054255 6.169336-10.066306M6.169336-10.066306Q6.224756-10.077498 6.283913-10.087384M6.283913-10.087384Q6.295412-10.089306 6.307069-10.091181M6.307069-10.091181Q6.352033-10.098414 6.399582-10.104994M6.399582-10.104994Q6.456619-10.112887 6.518082-10.119935M6.518082-10.119935Q6.57682-10.126671 6.640692-10.132758M6.640692-10.132758Q6.700214-10.138431 6.766135-10.143722M4.388293-8.821126Q4.399829-8.7978 4.41173-8.774369M4.41173-8.774369Q4.45334-8.692443 4.499166-8.609669M4.499166-8.609669Q4.543648-8.529323 4.591445-8.449364M4.591445-8.449364Q4.640969-8.366514 4.693366-8.285233M4.693366-8.285233Q4.749453-8.198227 4.808287-8.113867M4.808287-8.113867Q4.872417-8.021913 4.93946-7.933614M4.93946-7.933614Q4.975471-7.886185 5.012291-7.839854M5.012291-7.839854Q5.045116-7.798548 5.078582-7.758119M5.078582-7.758119Q5.090011-7.744311 5.881601-6.802823M5.881601-6.802823L6.664162-5.872074M6.664162-5.872074Q6.750616-5.769248 6.845624-5.673418M6.845624-5.673418L6.9589-5.559162M6.9589-5.559162L8.145691-6.463383M8.145691-6.463383Q8.217907-6.518405 8.27882-6.583845M8.27882-6.583845L8.825905-7.171586M8.825905-7.171586L9.102077-7.468282M9.102077-7.468282C9.241424-7.617984 9.382549-7.766027 9.520825-7.91673M9.520825-7.91673Q9.599154-8.002099 9.668271-8.081936M9.668271-8.081936Q9.70432-8.123577 9.738177-8.164068M9.738177-8.164068Q9.773035-8.205759 9.805768-8.246469M9.805768-8.246469Q9.839527-8.288456 9.871221-8.329645M9.871221-8.329645Q9.903971-8.372206 9.93471-8.414166M9.93471-8.414166Q9.96653-8.457604 9.996387-8.500659M9.996387-8.500659Q10.023246-8.539392 10.05587-8.590027M10.05587-8.590027Q10.085606-8.636182 10.113712-8.682708M10.113712-8.682708Q10.142701-8.730698 10.170128-8.779372M10.170128-8.779372Q10.198395-8.829536 10.225185-8.880749M10.225185-8.880749Q10.230051-8.890051 10.234872-8.899395M10.234872-8.899395Q10.257351-8.942962 10.27891-8.987548M10.27891-8.987548Q10.305728-9.043013 10.331304-9.100433M10.331304-9.100433Q10.337752-9.11491 10.344129-9.12953M10.344129-9.12953Q10.385597-9.22459 10.417641-9.309912M10.417641-9.309912Q10.420461-9.317421 10.423248-9.324953M10.423248-9.324953Q10.438711-9.36674 10.453201-9.409344M10.453201-9.409344Q10.47179-9.463999 10.488901-9.520365M10.488901-9.520365Q10.495469-9.541999 10.501831-9.563924M10.501831-9.563924Q10.508479-9.586835 10.51491-9.610086M10.51491-9.610086Q10.499723-9.621446 10.484562-9.632512M10.484562-9.632512Q10.429641-9.672601 10.374975-9.70889M10.374975-9.70889Q10.339958-9.732135 10.305-9.753853M10.305-9.753853Q10.287732-9.76458 10.270471-9.77494M10.270471-9.77494Q10.259619-9.781453 10.248768-9.787821M10.248768-9.787821Q10.210143-9.810491 10.171481-9.831358M10.171481-9.831358Q10.124935-9.856482 10.078198-9.879066M10.078198-9.879066Q10.034541-9.900162 9.990552-9.919123M9.990552-9.919123Q9.949629-9.936762 9.90824-9.95263M9.90824-9.95263Q9.869801-9.967368 9.830773-9.980649M9.830773-9.980649Q9.794505-9.992991 9.757537-10.00414M9.757537-10.00414Q9.723101-10.014527 9.687862-10.023937M9.687862-10.023937Q9.65491-10.032737 9.621065-10.040734M9.621065-10.040734Q9.589266-10.048249 9.556494-10.055098M9.556494-10.055098Q9.525541-10.061568 9.493542-10.067482M9.493542-10.067482Q9.434522-10.07839 9.370393-10.087667M9.370393-10.087667Q9.312205-10.096085 9.248043-10.10343M9.248043-10.10343Q9.189179-10.110168 9.123803-10.116173M9.123803-10.116173Q9.063411-10.12172 8.995683-10.126803M8.995683-10.126803Q8.987897-10.127387 8.979998-10.127967M8.979998-10.127967Q8.909298-10.133154 8.827194-10.138114',
  'M10.700016-9.680782Q10.815473-9.72382 10.925484-9.762058M10.925484-9.762058Q11.039948-9.801843 11.150062-9.836969M11.150062-9.836969Q11.263421-9.873131 11.373771-9.904864M11.373771-9.904864Q11.486287-9.93722 11.596843-9.965308M11.596843-9.965308Q11.652711-9.979502 11.708292-9.992663M11.708292-9.992663Q11.764119-10.005881 11.819804-10.018092M11.819804-10.018092Q11.875648-10.030337 11.9315-10.041602M11.9315-10.041602Q11.956776-10.0467 11.982078-10.051602M11.982078-10.051602Q12.01277-10.057548 12.043525-10.063211M12.043525-10.063211Q12.099612-10.073539 12.156044-10.082948M12.156044-10.082948Q12.191079-10.088789 12.226307-10.094287M12.226307-10.094287Q12.247734-10.097631 12.269247-10.10085M12.269247-10.10085Q12.325915-10.109329 12.383343-10.116966M12.383343-10.116966Q12.440438-10.124559 12.498562-10.131355M12.498562-10.131355Q12.540832-10.136297 12.583809-10.140838',
  'M7.15883-5.155221L7.245348-5.069074M7.245348-5.069074Q7.358689-4.956219 7.459938-4.833736M7.459938-4.833736L8.25756-3.868828M8.25756-3.868828L9.016953-2.950166M9.016953-2.950166C9.054733-2.904462 9 [... cut ...]
  var pathModel = makerjs.importer.fromSVGPathData(pathsArray[i]);
  var chain = makerjs.model.findSingleChain(pathModel);
  var count = Math.ceil(chain.pathLength / margin);
  var row;

  if(count > 1) {
    row = makerjs.layout.cloneToRow(rect, count, margin);
  } else {
    row = {models: { rect: makerjs.cloneObject(rect) } };
  }

  makerjs.layout.childrenOnChain(row, chain, 0.5, false, true);
  model.models.svgpaths.models[i]= pathModel;
  model.models.rows.models[i] = row;

}

module.exports = model;
```

closed 2023-01-28 by danmarshall

noobd3v, 2023-01-30:
@danmarshall  Thank You !!

noobd3v, 2023-01-30:
Is it part of version 0.17.3 ?

danmarshall, 2023-01-30:
just published in 0.17.4 on npm, the website is updating now.

noobd3v, 2023-01-31:
Okay Thanks a lot.

PR #555 (2023-01-28): handle layout for only one item
Fixes #549 
Fixes #550