microsoft/maker.js #550: childresnOnPath giving Nan

noobd3v, 2023-01-24:
Hi I was trying to generate children on path with following data

danmarshall, 2023-01-25:
Hi @noobd3v would you mind pasting the entire code? Thanks!

danmarshall, 2023-01-25:
Thank you very much for the code sample, it was helpful for the repro.
Looks like there is a guard missing for division by zero:

https://github.com/microsoft/maker.js/blob/master/packages/maker.js/src/core/layout.ts#L64-L66

This happens in a degenerate case when calling `layout.cloneTo...` with 1 as the count. Also, the core code should have a guard for the special case when count is <= 1. 

As a workaround for now, can you ensure your calling code only calls layout with a count > 1?
One more thing, probably good to pass a margin too:
```js
var row = makerjs.layout.cloneToRow(rect, 1);  //count is 1, no margin
```

PR #555 (2023-01-28): handle layout for only one item
Fixes #549 
Fixes #550
closed 2023-01-28 by danmarshall