microsoft/maker.js #554: draw model on center of line at any angle

Gautam2010, 2023-01-27:
How to draw model on center of line at any angle.  
It draws above the line.
[image]

Code I am trying is as below : 
`
 
    const totalDistance = makerjs.measure.pointDistance(line.origin, line.end);
    const halfDistance = totalDistance / 2 - moduleWidth / 2;
    const angle = makerjs.angle.ofLineInDegrees(line);
    makerjs.model.rotate(module, angle);
    const _angle = makerjs.angle.ofPointInRadians(line.origin, line.end);
    const x = line.origin[0] + halfDistance * Math.cos(_angle);
    const y = line.origin[1] + halfDistance * Math.sin(_angle);

    module.origin = [x, y];
`

Expectation is I am trying to make something below. Any idea what should be the correct approch?
[image]

MontassarAbdelhedi, 2023-01-27:
This can be wrong or not the case, but I feel it's worth sharing.
When you try to draw two shapes that occupy the same space, one of them gets pushed. I suggest doing this:

`var mainModel = { models: {}, paths: {} };
mainModel[0] = line;
mainModel[1] = module;
// calculate the position x and y
module.origin = [x, y]`

Gautam2010, 2023-01-30:
@MontassarAbdelhedi  It may works when you have no angle specified or simple case, Question is on when you have line with any angle.

danmarshall, 2023-01-30:
First you'll want to check the origin of the model you're rotating. Some models, like [Polygon](https://maker.js.org/playground/?script=Polygon) and [Star](https://maker.js.org/playground/?script=Star) are built around an origin at the center of their paths:
[image]
Others like [Rectangle](https://maker.js.org/playground/?script=Rectangle) and [Square](https://maker.js.org/playground/?script=Square) are built around an origin in the lower left:
[image]

If your model is symmetrical, you can convert between these modes with [zeroing and centering](https://maker.js.org/docs/intermediate-drawing/#Zeroing%20and%20Centering). If your model is not symmetric or you want an origin elswhere, you can try what @MontassarAbdelhedi suggested - "pushing" or "wrapping" your model inside another model. For example, with a Square with a side of 100:

```js
//this will have origin at lower left
var square = new makerjs.models.Square(100);
square.paths['center hole'] = new makerjs.paths.Circle([50,50], 2);

//offset the square to its center
square.origin = [-50, -50];

//rotate
makerjs.model.rotate(square, 45);

//can't set square's origin! - it needs to stay at [-50, -50] because that's where its children are referencing from.
//square.origin = [20, 0];	//bug

//so we "push" or "wrap" or "encapsulate" the rotated square into a model that we can move freely
var newModel = { 
  models: [ square ],

  //now we can move to the exact location we want
  origin: [20, 0]
};
```

Gautam2010, 2023-01-31:
Latest version `childrenOnPath`  fixed under #549

closed 2023-01-31 by Gautam2010