microsoft/maker.js #556: captions are not displayed in browser

MontassarAbdelhedi, 2023-01-30:
Hello,
I can't find a way to display the caption in the browser, this is my code:

`let makerjs = require("makerjs");`

danmarshall, 2023-01-30:
I get this svg:
```xml
<svg width="100" height="100" viewBox="0 0 100 100" xmlns="http://www.w3.org/2000/svg">
 <g id="svgGroup" stroke-linecap="round" fill-rule="evenodd" font-size="9pt" stroke="#000" stroke-width="0.25mm" fill="none" style="stroke:#000;stroke-width:0.25mm;fill:none"><path d="M 0 100 L 100 100 L 100 0 L 0 0 L 0 100 Z" vector-effect="non-scaling-stroke"/>
  <g id="captions">
   <text alignment-baseline="middle" text-anchor="middle" transform="rotate(0,50,50)" x="50" y="50">a square</text>
  </g>
 </g>
</svg>
```
It renders as a standalone svg file:
[image]

Can you use your browser's inspector to see if the caption element is present? Perhaps it's hidden by a CSS style.

danmarshall, 2023-01-30:
https://jsfiddle.net/z0k3ewgd/

MontassarAbdelhedi, 2023-02-01:
I tried with a new vue project. The question now is when moving a model with a caption, the caption doesn't move with it, is there a way to fix this?

danmarshall, 2023-02-01:
Oh, that's a bug! I have a fix now...

danmarshall, 2023-02-01:
Fixed in npm version 0.17.6

PR #558 (2023-02-01): Fix captions offset
Fixes #556