microsoft/maker.js #561: Module placement not showing correct angle

Gautam2010, 2023-03-02:
I am trying to draw model on chains using `layout.childrenOnChain` but some of modules are not placed correctly. 
Any idea on this?

[image]

Code reproduced here
https://codesandbox.io/s/makerjs-module-placement-angle-are-incorrect-mrrrwv

Gautam2010, 2023-03-15:
@danmarshall 
I was wondering if this fix is currently planned for a future sprint, or if it's still under consideration

danmarshall, 2023-03-15:
Hi @Gautam2010 I tried taking a look but ran out of time as it was a somewhat complicated scenario. I do remember in the code that there were 3 chains found. Knowing that the layout is based on [chainToPoints](https://maker.js.org/docs/working-with-chains/#Chain%20to%20points) I tried to just get the points then see that using [ConnectTheDots](https://maker.js.org/playground/?script=ConnectTheDots). I spent a bit of time with your example code but spent much time with the React / updating loop in CodeSandbox. Can you simplify the repro without using React?

noobd3v, 2023-03-29:
Hi @danmarshall  I was working on the issue and noticed that the way the angles are calculated for the rotation are dependent on the position of points with respect to the other position points which works but only when children on chain are closely placed.
https://github.com/microsoft/maker.js/blob/fb569bcb2501d73f8f31955f49b892033f5fcfcb/packages/maker.js/src/core/layout.ts#L154
In this case the points are far apart from each other hence the issue. 
It will be better if we can have the rotation angle set in respect with the path on which it is getting placed instead previous and next points.
[image]

noobd3v, 2023-03-29:
There could be multiple ways of updating
1. Could be to take argument for distance of previous and next point just for rotation angle calculation. In a way baseline for mitered angle previous next points.
2. Take the path segment at the position to get the angle.

I guess best would be to use the path segment for the placement point to get the angle.

danmarshall, 2023-03-29:
@noobd3v thanks for working on the issue 👍🏽 and you are correct that is the method for layout. I agree that the method you suggest may also be useful. There's scenarios where each style might be appropriate, so this could be a new option.

noobd3v, 2023-04-10:
Hi @danmarshall  I tried updating the existing function and added the extra parameter to take as flag for rotation along the placed path.
I used following approach.
1. So that we don't have to traverse the whole chain to get the path for the cpa point I updated the toPoints method of chain to provide path context of the point (it can be helpful in some other applications of toPoints so created an overload with callback similar to what is done with findChains).
2. Used the path context to get the angle made by endpoints.
3. Assigned the angle for rotation.
But it is not providing the required results. One thing that's not clear to me was the points that are being generated, below is the screenshot of the points with respect to the path and the points are not on the path.
[image]

Here are the changes for your reference that I have done to achieve the functionality. Can you please look into it let me know  where I am wrong about my approach.
https://github.com/microsoft/maker.js/compare/master...noobd3v:maker.js:rotation-along-path

noobd3v, 2023-04-10:
I might be providing more than enough information in the callback but I thought it might be useful in some cases.

noobd3v, 2023-04-10:
Got what I was doing wrong.
I was using the 1st index of the points which was the start and should be avoided.
And the points were the mirror of the actual shape.
[image]

noobd3v, 2023-04-10:
Created PR for feature.
https://github.com/microsoft/maker.js/pull/566