microsoft/maker.js #562: The byLayers option is not working properly in findChains

techa, 2023-03-06:
Setting the layer name to the model does not work because the layer name is taken from the path [here](https://github.com/microsoft/maker.js/blob/master/packages/maker.js/src/core/chain.ts#L169).

Alternative solution.
- Use `model.layer()` to reflect layer names on child models and paths as well.

danmarshall, 2023-03-06:
A layer on a path will override a layer on the containing model.

techa, 2023-03-07:
Sorry, I tried it with a simple code and you were right.

However, with my code, this does not work as expected.
```
model.layer = 'layer-name';
```

If I use a function that overwrites the layer name to lower paths, it works as expected.

There must be something else causing this. I will try to find it.
Thanks for the reply.

closed 2023-03-08 by techa

techa, 2023-03-08:
It was caused by layer names remaining in the child models of the model created by compositing with `model.combine()`.