microsoft/maker.js #563: Is there any way to use reference element

Gautam2010, 2023-03-06:
Checking with makerjs how can we use `useElement`?
https://www.w3.org/TR/SVG/struct.html#UseElement

danmarshall, 2023-03-06:
Is this for a repeating background? You may need to manipulate the svg.

Gautam2010, 2023-03-07:
With below image red marked are big svg path model, 
I want to place 1 original svg with path and reuse all other model with `<use href="#original-module-id" />`

I have tried in Makerjs

`const refElement = new makerjs.exporter.XmlTag("use", {
          href: "#original-module-id",
});`

but how can I include `refElement` into my model?
I am using `makerjs.layout.cloneToRow(refModel, numberOfModuleCanPlace)` to place module. How we can use element insead of refModel

[image]

Gautam2010, 2023-03-15:
@danmarshall 
Hi there! I was wondering if it has a feature similar to the one described in my previous comment. Can you please let me know if this is something that your library supports? Thanks!

danmarshall, 2023-03-15:
Hi @Gautam2010 it's not supported. You may want to use chain.ToPoints and layout these with your own code using the [algorithm in the source code](https://github.com/microsoft/maker.js/blob/master/packages/maker.js/src/core/layout.ts#L173).

Gautam2010, 2023-03-22:
@danmarshall 
I am trying to implement using points. 
Is there any way to attach `makerjs.exporter.XmlTag` to interface `IModel`

danmarshall, 2023-03-22:
> Is there any way to attach `makerjs.exporter.XmlTag` to interface `IModel` 

Can you explain more about this? I'm not sure I understand.

Gautam2010, 2023-03-22:
@danmarshall 
`IModel` interface I can see below properties for my final prepared model, where I can add all the path and models.
```
interface IModel {
        origin?: IPoint;
        "type"?: string;
         paths?: IPathMap;
        models?: IModelMap;      
        units?: string;
        notes?: string;
        layer?: string;
        caption?: ICaption;
        exporterOptions?: { [exporterName: string]: any };
    }
```
 
I have created `const refElement = new makerjs.exporter.XmlTag("use", { href: "#original-module-id", });` element, but I am not sure what property is to assign in `IModel` since it is not type of `path `or `models`

Is there any way I can add `XmlTag` to `IModel`

something below:

```
const refElement = new makerjs.exporter.XmlTag("use", { href: "#original-module-id", });
var finalModel:IModel {
      element : {
          'xml-tag-element-1':refElement 
     }
}
```

danmarshall, 2023-03-22:
You can extend the interface for your code:
```ts
interface MyModel extends IModel {
  myProp: any;
}
```

Gautam2010, 2023-04-19:
I was able fix with `chain.ToPoints` and override code 
Thanks for the help Marking this as closed.

closed 2023-04-19 by Gautam2010