microsoft/maker.js #567: Exception While getting outline for the font model

Gautam2010, 2023-04-24:
Throwing  error while getting outline from font model.

[image]

With debug `otherElement` getting undefined and `this.removeValue(otherElement, valueId);` method is using `valueIds` of `otherElement` which undefined.

[image]

Gautam2010, 2023-04-25:
Hi @danmarshall 

Here is the [link](https://codesandbox.io/p/sandbox/y96qjy?file=%2Findex.js) to reproduce the error.
In above link code I am trying to generate fontModel as per below

[image]

Along with it also want to draw inner outline with distance 1. Like below done in char "m".

[image]

Gautam2010, 2023-05-05:
Hi @danmarshall,

Just wanted to check in and see if there has been any progress on fixing this bug. Is there a plan to address it in the coming week? Let me know if there's anything I can do to help. Thanks!

danmarshall, 2023-05-05:
Hi @Gautam2010 - sorry I will not be addressing this directly, and not soon as I am busy with other projects. This is probably an instance of the [prevalent bug in boolean operations](https://github.com/microsoft/maker.js/issues?q=is%3Aissue+is%3Aopen+boolean). I think the solution to this will be foundational (meaning a lot of other low level stuff will need to change and be tested in order to get this right). In the meanwhile, you may try modifying your outline distance by tiny amounts to see if you can get a specific instance working properly.