microsoft/maker.js #569: Not Indexing full svg path while making model fromSVGPathData

Gautam2010, 2023-09-01:
Trying to make `model` with simple below svg path
```
M249.286 106.372L471.068 327.86 
H471.068 27.505 27.505 L249.286 106.37200000000001Z
```
This is the preview of svg path.
[image]

I am making model with bolow code.
```
const svgPath=`M249.286 106.372L471.068 327.86 
H471.068 27.505 27.505 L249.286 106.37200000000001Z`

const makerModel=makerjs.importer.fromSVGPathData(svgPath);
```

Due to some reason it is not indexing some of lines.

`const svg=makerjs.exporter.toSVG(makerModel);`

Response of final svg becomes like below
[image]

Final svg output
`<svg width="221.782" height="221.488" viewBox="0 0 221.782 221.488" xmlns="http://www.w3.org/2000/svg"><g id="svgGroup" stroke-linecap="round" fill-rule="evenodd" font-size="9pt" stroke="#000" stroke-width="0.25mm" fill="none" style="stroke:#000;stroke-width:0.25mm;fill:none"><path d="M 0 0 L 221.782 221.488 L 0 0 Z" vector-effect="non-scaling-stroke"/></g></svg>`

**It should give same output as input.**

[Code reproduced link](https://codesandbox.io/p/sandbox/clever-wing-wwm3dt?file=/index.js:9,18)

Gautam2010, 2023-09-11:
@danmarshall 
Is this something that MakerJs is designed to handle, or could it potentially be considered for inclusion in a future release?

Your insights would be greatly appreciated. Thank you!

danmarshall, 2023-09-11:
The [parser code](https://github.com/microsoft/maker.js/blob/master/packages/maker.js/src/core/svg.ts#L983) probably isn't robust enough to handle this case. PRs are welcome.

Meanwhile, if you check, *then uncheck* the "Minify output" button in the [svg path inspector](https://yqnn.github.io/svg-path-editor/) you can get a standardized output:
```
M 249.286 106.372 L 471.068 327.86 H 471.068 H 27.505 H 27.505 L 249.286 106.372 Z
```