microsoft/maker.js #581: How can the relative coordinates of all the text paths in the rectangles be obtained in makerjs?

PPDEMOone, 2023-11-14:
How can the relative coordinates of all the text paths in the rectangles be obtained in makerjs?

1. Firstly I use the (makerJs.importer.fromSVGPathData) method and create the path, and  I got the textModelnstance from the (makerJs.models.Text)
2. Then I moved the origin of both model objects to [0,0] with (makerjs.model.zero)
3. I use the (makerjs.model.walkPath) to get  all the path origin in the textModelnstance

But the  textModelnstance have some Bessel curves and paths, I'm not sure which parameters I want in (makerjs.model.walkPath) method result

[image]

PPDEMOone, 2023-11-14:
I just found this method （makerJs.model.originate） seems to solve my problem

```typescript
const model: IModel = {
  models: {
    bc3: bottomModel,
    text: textModel,
    measure: measureRect,
    ref_line_1: createDashedLine([0, 0], bModelTopEndPointRight!, 5),
    ref_line_2: createDashedLine([0, 0], bModelTopEndPointLeft!, 5),
    ref_middle_line: createDashedLine(
      [0, measureRectCenter[1]],
      [measureRectHigh[0], measureRectCenter[1]],
      5
    ),
    ref_bottom_line: createDashedLine(
      [0, measureBottomHigh[1]],
      [measureRectHigh[0], measureBottomHigh[1]],
      5
    ),
  },
  paths: {
    line: new makerJs.paths.Line([0, 0], [202.3195951860165, 81.1357899049327]),
  },
};

makerJs.model.originate(model);

makerJs.model.walkPaths(textModel, (val) => {
  if (val.type === "BezierCurve") {
    const bezierCurvePath = val as InstanceType<
      typeof makerJs.models.BezierCurve
    >;
    console.log(bezierCurvePath);
  } else {
    // console.log(val.origin);
  }
});
``` 

[image]

PPDEMOone, 2023-11-16:
> I just found this method(makerJs.model.originate)seems to solve my problem
> 
> ```ts
> const model: IModel = {
>   models: {
>     bc3: bottomModel,
>     text: textModel,
>     measure: measureRect,
>     ref_line_1: createDashedLine([0, 0], bModelTopEndPointRight!, 5),
>     ref_line_2: createDashedLine([0, 0], bModelTopEndPointLeft!, 5),
>     ref_middle_line: createDashedLine(
>       [0, measureRectCenter[1]],
>       [measureRectHigh[0], measureRectCenter[1]],
>       5
>     ),
>     ref_bottom_line: createDashedLine(
>       [0, measureBottomHigh[1]],
>       [measureRectHigh[0], measureBottomHigh[1]],
>       5
>     ),
>   },
>   paths: {
>     line: new makerJs.paths.Line([0, 0], [202.3195951860165, 81.1357899049327]),
>   },
> };
> 
> makerJs.model.originate(model);
> 
> makerJs.model.walkPaths(textModel, (val) => {
>   if (val.type === "BezierCurve") {
>     const bezierCurvePath = val as InstanceType<
>       typeof makerJs.models.BezierCurve
>     >;
>     console.log(bezierCurvePath);
>   } else {
>     // console.log(val.origin);
>   }
> });
> ```
> 
> [image]

closed 2023-11-16 by PPDEMOone

danmarshall, 2023-11-16:
Thanks for the question. For anyone else who wants to know more please see [originating](https://maker.js.org/docs/intermediate-drawing/#Originating) in the documentation.