microsoft/maker.js #582: exports svg path data breaking when fill color

PPDEMOone, 2024-01-09:
I am attempting to use the `toSVG` method to export my processed model in my code, however, the exported result is unsatisfactory under certain conditions.

I use the svgPathData in the online SVG viewer. If I only stroke this SVG path,  it appears normal

[image]

But when I fill this svg... :(

[image]

What might be causing this
That's my demo
[](https://stackblitz.com/~/github.com/PPDEMOone/maker-convert)

PPDEMOone, 2024-01-09:
There are some broken paths in the SVG that has been generated.

danmarshall, 2024-01-09:
Hello, cool app! Something in the `while` loop seems to cause it to export. When I removed your code:
```ts
while (
  targetMeasureRect.width >= CARVED_MAX_WIDTH ||
  targetMeasureRect.height >= CARVED_MAX_HEIGHT ||
  !isContain(collisionRegionRect, targetMeasureRect.low, targetMeasureRect.high)
) {
  if (times) {
    scale = Number((scale - SCALE_RATIO_STEP).toFixed(2));
  }

  if (scale <= 0) {
    break;
  }

  lineTextModel = lineTextContent.models!["lineTextModel"] =
    makerJs.model.distort(lineTextModel, scale, scale);
  targetMeasureRect = makerJs.measure.modelExtents(lineTextModel);
  // console.log(targetMeasureRect);
  makerJs.model.moveRelative(lineTextModel, [
    (CARVED_MAX_WIDTH - targetMeasureRect.width) / 2,
    (CARVED_MAX_HEIGHT - targetMeasureRect.height) / 2,
  ]);

  const { low, high } = targetMeasureRect;
  const overflowBottom = makerJs.measure.isPointInsideModel(
    low,
    collisionRegionRect
  );
  const overflowTop = makerJs.measure.isPointInsideModel(
    high,
    collisionRegionRect
  );

  if (overflowBottom && overflowTop) {
    break;
  } else if (!overflowTop) {
    makerJs.model.moveRelative(lineTextModel, [0, -Math.abs(high[1])]);
  } else if (!overflowBottom) {
    makerJs.model.moveRelative(lineTextModel, [0, Math.abs(low[1])]);
  } else {
    continue;
  }
  targetMeasureRect = makerJs.measure.modelExtents(lineTextModel);
  times++;
   console.log(scale);
}
```
I was able to get a better export:
[image]
 
 Looking at this code, I'm guessing the `distort` function call is where the bug might be occurring. Perhaps some unit tests on distorting a font/text might help encapsulate the condition.

PPDEMOone, 2024-01-10:
> Hello, cool app! Something in the `while` loop seems to cause it to export. When I removed your code:
> 
> ```ts
> while (
>   targetMeasureRect.width >= CARVED_MAX_WIDTH ||
>   targetMeasureRect.height >= CARVED_MAX_HEIGHT ||
>   !isContain(collisionRegionRect, targetMeasureRect.low, targetMeasureRect.high)
> ) {
>   if (times) {
>     scale = Number((scale - SCALE_RATIO_STEP).toFixed(2));
>   }
> 
>   if (scale <= 0) {
>     break;
>   }
> 
>   lineTextModel = lineTextContent.models!["lineTextModel"] =
>     makerJs.model.distort(lineTextModel, scale, scale);
>   targetMeasureRect = makerJs.measure.modelExtents(lineTextModel);
>   // console.log(targetMeasureRect);
>   makerJs.model.moveRelative(lineTextModel, [
>     (CARVED_MAX_WIDTH - targetMeasureRect.width) / 2,
>     (CARVED_MAX_HEIGHT - targetMeasureRect.height) / 2,
>   ]);
> 
>   const { low, high } = targetMeasureRect;
>   const overflowBottom = makerJs.measure.isPointInsideModel(
>     low,
>     collisionRegionRect
>   );
>   const overflowTop = makerJs.measure.isPointInsideModel(
>     high,
>     collisionRegionRect
>   );
> 
>   if (overflowBottom && overflowTop) {
>     break;
>   } else if (!overflowTop) {
>     makerJs.model.moveRelative(lineTextModel, [0, -Math.abs(high[1])]);
>   } else if (!overflowBottom) {
>     makerJs.model.moveRelative(lineTextModel, [0, Math.abs(low[1])]);
>   } else {
>     continue;
>   }
>   targetMeasureRect = makerJs.measure.modelExtents(lineTextModel);
>   times++;
>    console.log(scale);
> }
> ```
> 
> I was able to get a better export: <img alt="image" width="938" src="https://private-user-images.githubusercontent.com/11507384/295350789-6035e68e-d804-486b-905f-7cd51a4cb784.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDQ4NDk2NjcsIm5iZiI6MTcwNDg0OTM2NywicGF0aCI6Ii8xMTUwNzM4NC8yOTUzNTA3ODktNjAzNWU2OGUtZDgwNC00ODZiLTkwNWYtN2NkNTFhNGNiNzg0LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDAxMTAlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwMTEwVDAxMTYwN1omWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWMxYjI2NThhOWZiYzIwOTM5NDdkNzhjYTM5YzQyZmUzNDcyZmJiYjBiNDUyNDgyYzhlNTQ2YTdhN2VhZWE3MGQmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.LQ28DavJ2uQ9GJ7G03Dl4cggSi3siKF9ko-OnpTHqq8">
> 
> Looking at this code, I'm guessing the `distort` function call is where the bug might be occurring. Perhaps some unit tests on distorting a font/text might help encapsulate the condition.

PPDEMOone, 2024-01-10:
As in the above case, the `makerjs.models.scale` function happens, too 🤔

PPDEMOone, 2024-01-10:
when I give a default distort value, it‘s normally working
```typescript
lineTextModel = lineTextContent.models!['lineTextModel'] = makerJs.model.distort(
      lineTextModel,
      0.9,
      0.9,
    );
    targetMeasureRect = makerJs.measure.modelExtents(lineTextModel);
    makerJs.model.moveRelative(lineTextModel, [
      (CARVED_MAX_WIDTH - targetMeasureRect.width) / 2,
      (CARVED_MAX_HEIGHT - targetMeasureRect.height) / 2,
]);
``` 
[image]

There's no break in this path 🧐

PPDEMOone, 2024-01-10:
Also, I call the `scale` method and export the svgPathData, some points are  not connected
```typescript
makerJs.model.zero(guitarKnife);
makerJs.model.originate(guitarKnife);
makerJs.model.scale(guitarLineTextModel, 0.3, false);
makerJs.model.addModel(guitarKnife, guitarLineTextModel, "guitarLineTextModel");
console.log(makerJs.exporter.toSVGPathData(guitarKnife));
``` 
[image]

The right side is normal and the left side is not connected

[image]

I read parts `scale` and `distort` methods in the source code.
I believe that the issue is being caused by the export rules for the SVG file format, I am currently still working on that.

closed 2024-01-18 by PPDEMOone