microsoft/maker.js #604: How can I get the angle of the text created by model.Text()?

doox911-opensource, 2024-08-20:
```js
export function getTextModel(font, caption, font_size) {
  const model = new makerjs.models.Text(font, caption, font_size);

  model.layer = DrawLayerNames.Caption;

  const model_extents = makerjs.measure.modelExtents(model);

  return [model, model_extents];
}

const [caption, caption_extents] = getTextModel(font, `H: ${H}`, render_options.font_size);

  H_caption.origin = [
  // some origin/end
  ];

  makerjs.model.rotate(H_caption, /* some angle*/, H_caption.origin);
```

How can I find out how rotated the text is in another part of the program?

danmarshall, 2024-08-20:
Hello, one thing to note is that your drawing is a plain object to which you can add your own properties:

```js
caption.rememberMyRotation = 55;
```

Properties not used by Maker.js will remain untouched. Hope that helps.

doox911-opensource, 2024-08-22:
> Hello, one thing to note is that your drawing is a plain object to which you can add your own properties:
> 
> ```js
> caption.rememberMyRotation = 55;
> ```
> 
> Properties not used by Maker.js will remain untouched. Hope that helps.

Ohhhh(

This decision was one of the first in my head) But I thought that maybe someone solved the problem in a different way.

My attempt to solve the problem without adding my own properties:

```js
export function getCircumscribedRectangleAngle(model) {
  const { center, width, height, high } = makerjs.measure.modelExtents(model);

  if (width >= height) {
    const rightMiddle = [high[0], center[1]];

    return makerjs.angle.ofPointInDegrees(center, rightMiddle);
  } else {
    const topMiddle = [center[0], high[1]];

    return makerjs.angle.ofPointInDegrees(center, topMiddle);
  }
}
```

Has the development of the library stopped? A similar method would be very useful)

doox911-opensource, 2024-09-03:
[image]

[image]

I figured out why it doesn't work with slanted lines) Since the makers.measure.model Extends method returns the described rectangle, there are only 4 possible angles.