microsoft/maker.js #615: [Bug] Invalid value in the method makerjs.measure.pointDistance

doox911-opensource, 2024-11-19:
Invalid value (six characters after the dot):

```js
makerjs.measure.pointDistance([100, 100], [100, 99.999999]) // 9.999999974752427e-7
```

valid value (five characters after the dot):

```js
makerjs.measure.pointDistance([100, 100], [100, 99.99999]) // 0.000010000000003174137
```

z3dev, 2024-11-19:
What does this mean? Numeric imprecision is a know problem in all algorithms. What's the use case?

doox911-opensource, 2024-11-20:
> What does this mean? Numeric imprecision is a know problem in all algorithms. What's the use case?

You're right, this is not a bug(It's JS). 

Now I found `round` in makerjs.

Here is an example of usage:

```js
function roundCoordinate(coordinate) {
  return makerjs.round(coordinate, 0.001);
}

function roundPointCoordinates(point) {
  return [point[0], point[1]].map(coordinate => roundCoordinate(coordinate));
}

function isTooClose(point1, point2, attraction_length = 0.001) {
  const p1 = roundPointCoordinates(point1);

  const p2 = roundPointCoordinates(point2);

  return makerjs.measure.pointDistance(p1, p2) < attraction_length;
}
```

closed 2024-11-20 by doox911-opensource