microsoft/maker.js #622: Unifying beziers curves when finding chains returns some arcs in pathContext instead of the bezier seed paths

ggoodkey, 2025-05-12:
I am expecting that calling model.findChains with the unifyBeziers option set to true, should return the bezier seeds instead of the individual arcs. However, some links (every other?) in the chain contain the last arc of the curve as the returned pathContext instead. This seems like a bug. The seed is still available by digging through the returned modelContext as a work around

danmarshall, 2025-05-12:
Thanks for posting, can you provide a small repro?

ggoodkey, 2025-05-13:
[https://github.com/ggoodkey/unify-bezier](https://github.com/ggoodkey/unify-bezier)

or in the playground
```javascript 
var makerjs = require('makerjs');

this.models = {};
this.models.curve = {
  models:{
    curve1: new makerjs.models.BezierCurve([0, 0], [25, 35], [30, -10], [50, 0]),
    curve2: new makerjs.models.BezierCurve([50, 0], [75, 35], [80, -10], [100, 0])
  }
};

var text = "";
makerjs.model.findChains(this.models.curve, function (chains, loose, layer, ignored) {
  for (var a = 0; a < chains.length; a++) {
    for (var b = 0; b < chains[a].links.length; b++) {
      text = chains[a].links[b].walkedPath.pathContext.type;
      this.models['text_' + b] = {};
      makerjs.model.addCaption(this.models['text_' + b], text, [0, -10 * (b + 1)], [100, -10 * (b + 1)]);
    }
  }
}.bind(this), { unifyBeziers: true });
```

ggoodkey, 2025-05-23:
Just working on this problem a bit more. I seem to have tracked the issue down to a comparison between 2 numbers at Line 258 in the Bezier Curve model, which are always almost, but not quite equal, i.e. 1.2345678 vs 1.23456999996778.

Line 258 --> `if (!measure.isPointEqual(ends[i], tPoints[i].point)) {`

`isPointEqual` rounds off to .0000001 by default, but if I set it to .000001, the problem goes away.

`if (!measure.isPointEqual(ends[i], tPoints[i].point, .000001)) {`

A few thoughts:
The code is comparing the end points of the curve segment with the calculated point on the curve. If that margin of error can be improved elsewhere in the code, maybe when the Bezier Curve is created, that might be the proper solution, but maybe it's a Bezier.js problem?

Also, increasing the accuracy of the Bezier Curve, even to a ridiculous amount that takes ages to compute, doesn't seem to have any impact on this comparison.

danmarshall, 2025-05-24:
Thanks for looking into this! There is a [pointMatchingDistance property of IFindChainsOptions](https://maker.js.org/docs/api/interfaces/makerjs.ifindchainsoptions.html#pointmatchingdistance) but it seems its not passed down through to that comparison.

danmarshall, 2025-05-24:
Thanks so much for digging in, this was one issue that I had been peeved about for a long time in https://danmarshall.github.io/google-font-to-svg-path/  - I always saw arcs in the svg when I knew there shouldn't be any.

PR #623 (2025-05-24): Add pointMatchingDistance parameter to bezier seed functions
fixes #622