microsoft/maker.js #627: Translating an entire model doesn’t translate its nested children

deusprogrammer, 2025-06-27:
I would really like to be able to translate a whole model and its children, but it doesn’t seem to work in practice.  I don’t know if this just isn’t supported or if it doesn’t work.  If it’s not supported I will fork the repo and add it.

danmarshall, 2025-06-28:
Glad you got a handle on it. And now I'm intrigued about Taco Truck 🌮

deusprogrammer, 2025-06-28:
I would be happy to show you.  I make custom fightsticks, so I made a tool for creating custom panel layouts.  It uses Pixi.js for the UI, and I export SVG's out of it that are ready for the laser cutter.  But people wanted DXF exports to send to SendCutSend, so I found out about your project.  I am going to rewrite Taco Truck to store everything in the JSON format Maker.js uses (instead of my proprietary one).  This way my project can have a single source of truth that can export SVG, DXF, and I can just rewrite the Pixi.js rendering code to use the same format.

Here's an example of a more complex layout:

https://cg-taco-truck.web.app/designer/projects/xe0HmYMRzdnkBqU0Hmh9

deusprogrammer, 2025-06-30:
That was ultimately what I was thinking of doing.  I managed to write something that will take an arbitrary SVG and render it using pixijs’s graphics commands and then next I will write something to compile it into makerjs’s json format.

closed 2025-06-30 by danmarshall

deusprogrammer, 2025-07-02:
I'm really sorry to keep bothering you sir, but is there any possibility I can get in touch with you outside of github?  I want to pick your brain on somethings.

If you are Discord, would you be able to reach out to me at thetruekingofspace?

Please don't feel pressured or obliged to do so of course as I'm sure you are a busy man.

danmarshall, 2025-07-04:
Any chance that you can ask SendCutSend to give a specific DXF error?